robotology-legacy/icub-gazebo-legacy #15: iCub force balancing working smoothly in Gazebo simulation

traversaro, 2015-03-19:
I open this meta issue (that is not only related to `gazebo-yarp-plugins`) to collect in one place links to all the issue related to the problem of running the iCub force  balancing (using the `torqueBalancing` module available in the `codyco-modules` repository) in Gazebo, for helping people that are going to work on this: @jeljaik @naveenoid @prashanthr05 @nunoguedelha . 

As we discussed, there are several fix that where directly committed to this repository without updating the model in it. For an overview on how the model in this repo are generated, please check this issue: https://github.com/robotology/codyco-superbuild/issues/55 .  

Users from other partners of the CoDyCo consortium are having troubles in dealing with iCub contacts: https://github.com/robotology/gazebo-yarp-plugins/issues/178 . 

There is a open issue related to synchronization of the simulation with YARP software : https://github.com/robotology/gazebo-yarp-plugins/issues/173 . 

Gazebo supports several physics engines, but so far we just ran the iCub-related simulations in ODE: https://github.com/robotology/gazebo-yarp-plugins/issues/146 https://github.com/robotology/gazebo-yarp-plugins/issues/126 https://github.com/robotology/gazebo-yarp-plugins/issues/128 . 

We currently were using the `SDF` model for the robot, because FT sensor are modeled in URDF and SDF as fixed joints, but fixed joints in former versions of Gazebo (before libsdformat 2.3) where automatically removed. Now an option to disable this "fixed joint lumping" has been added to Gazebo https://bitbucket.org/osrf/gazebo/issue/618/add-option-to-disable-auto-fixed-joint https://bitbucket.org/osrf/gazebo/issue/618/add-option-to-disable-auto-fixed-joint , so we can now try to load directly the URDF generated by `icub-model-generator` in Gazebo (by properly adding some configuration tags in the URDF). This will simplify thinks because all our modules (`torqueBalancing`, `wholeBodyDynamicsTree`, etc etc) load the robot model from URDF. In this case we can be absolutely sure that the model used in the modules and in the simulator is exactly the same.

nunoguedelha, 2015-03-24:
Prashanth and I have implemented the replacement of the feet mesh by a simpler shape in the iCub SDF model. As per the team recommendations, we've used a simple box shape leading to a 4 contact/collision points with the ground. This is under test and being tracked under a previously open issue:
https://github.com/jeljaik/WB-Team-Issues/issues/27

traversaro, 2015-03-24:
@jeljaik the two procedures currently used for generating urdf models are described in this issue: https://github.com/robotology/codyco-superbuild/issues/55 .

traversaro, 2016-11-08:
This is working fine now (see https://www.youtube.com/watch?v=UXU3KSa201o).
Thanks to everyone involved @jeljaik @prashanthr05 @nunoguedelha !

closed 2016-11-08 by traversaro