robotology-legacy/icub-gazebo-legacy #16: Give Gazebo models proper YARP_ROBOT_NAME

traversaro, 2015-04-27:
Currently we are storing gazebo-specific configuration under the robot `icubGazeboSim`. However, we now have two different models of the robot in Gazebo [1] :
* A model similar to an iCub v1, generated using the old generation system in https://github.com/robotology-playground/icub-model-generator . 
* A model of the `iCubHeidelberg01`, generated using directly from the Creo CAD models, as described in http://wiki.icub.org/wiki/Creo_Mechanism_to_URDF . 

Hopefully one day we will have also a proper model of iCub v2.5 generated from Creo CAD models, and eventually also a iCub 3 model: we should decide a naming schema for their `YARP_ROBOT_NAME`. 

For the Gazebo version of `iCubHeidelberg01`, I propose `iCubHeidelbergGazebo` . 
However I can't decide on the other models, on possible choice could be:
`iCubGazebo01` for iCub v1 ,
`iCubGazebo02` for iCub v2.5 model, 
and so on. However I don't like this because it is not clear that the 01/02 is the version of iCub, and not a simple serial number as in `iCubGenova01/02/03` .
Another possibility is:
`iCubV1Gazebo`,
`iCubV2Gazebo`,
`iCubV3Gazebo`.
That I like more, but is a bit ugly. 
Any opinions @DanielePucci @jeljaik ? 

[1] :  https://github.com/robotology/codyco-superbuild/issues/55 .

rlober, 2015-04-27:
Why not `iCubGazeboV1`, `iCubGazeboV2` ... and for local models extracted from cad `iCubGazeboParis01` or `iCubGazeboParis02`.

Just my two cents :smile:

traversaro, 2015-04-27:
I like `iCubGazeboV1` and `iCubGazeboV2`.

DanielePucci, 2015-04-27:
It's fine for me also: I like iCubGazeboV1 and iCubGazeboV2 .

jeljaik, 2015-04-27:
I think it's important also to differentiate between iCubV2 and iCubV2.5 for example ... we might simply add it at the end as `iCubGazeboV2_5` to at least have a clear relationship with the 'real' version of the robot it's simulating. I know it's a bit ugly, but some easy mapping must be done among existing names/versions and these new ones. `iCubGenova01` is the perfect example for this, being I think the only robot with its characteristics (neck version, -almost- full body skin, skin under the feet, legs v2.5, etc). Maybe the current changes might not mean much at the simulation level, but if tomorrow they add the battery pack to `iCubGenova01` for example, that would change a bit things wouldn't it? in terms of total mass, center of mass and so on. Would it be fair to call it v2.5?

traversaro, 2015-04-27:
Let's clarify: I don't want to maintain a "Gazebo" version for all exiting robots, I think this would be difficult to maintain. 

I think for all practical use it is ok to just maintain: 
* Heidelberg version, given that is completely different
* A model for each major version of the iCub (i.e. 1, 2, 3).

Having say that, I thinks it could make sense to clearly distinguish iCub v2 from v2.5, even in the `YARP_ROBOT_NAME`, but I don't think we will ever have a proper iCub v2 in simulation (given that it is quite similar to v1, except for ankle motors.

traversaro, 2022-01-13:
In the end, we used indeed `iCubGazeboV3` style names, so I think we can close this issue.

closed 2022-01-13 by traversaro