robotology-legacy/icub-gazebo-legacy #19: [iCub] wrong leg limits

francesco-romano, 2015-07-17:
It seems that the min and max limits are exchanged (at least for the legs)

traversaro, 2015-07-17:
For models generated with the "old" generation procedure, limits are copied from the urdf files provided by UPMC :  https://github.com/robotology-playground/icub-model-generator/blob/master/src/icub_urdf_sdf_generator.cpp#L679

We should then modify the source files : 
 * https://github.com/robotology-playground/icub-model-generator/blob/master/data/urdf_paris/icub_simple/icub_simple.xml
 * https://github.com/robotology-playground/icub-model-generator/blob/master/data/urdf_paris/icub/icub.xml 
or get the limits from some other source, and then regenerate all the files.

traversaro, 2015-07-17:
Limits issue originally tracked in https://github.com/robotology/codyco-superbuild/issues/55 .

traversaro, 2016-11-08:
@francesco-romano You think we fixed this?

francesco-romano, 2016-11-08:
Honestly, I don't remember.
Let's close this for now, and reopen it in case it hasn't been fixed yet.

closed 2016-11-08 by francesco-romano