robotology-legacy/icub-gazebo-legacy #21: enabling self collision

rlober, 2015-08-11:
Where should I put the,
```xml
<selfCollide>true</selfCollide>
```
tag in the icub models to enable self collision? I would like to do it for the whole body if possible.

I haven't been able to find a clear answer on this in the google-sphere.

Thanks
Ryan

traversaro, 2015-08-11:
The tag for self collision in SDF is `self_collide`, and is a child of `model` tag. The official documentation should be: http://sdformat.org/spec?ver=1.5&elem=model#model_self_collide . Anyway it could make sense we enable it by default...

rlober, 2015-08-11:
Hey @traversaro, thanks for the help. I will play around with it a bit then come back here and report my experience. With other simulators enabling self collision tends to drastically increase computational load so I am wary of enabling be default, but I will let you know when I have more experience with it. I am going to close this issue for now. Thanks man!

rlober, 2015-08-12:
@traversaro 

So I have played a bit with `self_collide` and found some issues. Just FYI, I am running Gazebo 4.1.1 and currently have version 2.2.0-1~precise of `libsdformat2`.

Issues:
- The `<self_collide>` element is not a child of `<model>` but rather a child of `<link>`. I think this is an unresolved issue for them: https://bitbucket.org/osrf/sdformat/issues/71

- I first tried adding `<self_collide>` to all links... Not a good idea. This causes robot epilepsy. The problem has to do with links constantly being in collision with sibling links.

- If I just enable `<self_collide>` for the hands and forearms I get more or less the behavior I am looking for but there is still some unwanted collision between the hands and the wrists:
[image]

- My thought is that if we modify the hand collision meshes to not include the base of the hands then maybe this will be resolved. I tried to open the collision mesh for the hand in blender but nothing showed up so I don't know where to go from here. What do you guys use to edit your meshes?

traversaro, 2015-08-13:
My tests : 
 * On Gazebo 6 , sdformat 3 adding a `<self_collide>true</self_collide>` tag to the model was sufficient to enable self collision on all the links. I don't know if this is due to a bug in Gazebo 4 that is now fixed in Gazebo 6. However by doing this I get the problem that you where experiencing of nearby links colliding. (Brief comments on this: Gazebo disable collision for two links connected by a joint, but for some links (such as `l_forearm` and `l_hand` we have two "collision" links that are connected by two joints, so the collision between them are not disabled). 

* For editing meshes, I had some limited experience with MeshLab, but I don't know if it is the right tool to modify the shape of a mesh. I guess Collada (dae) is not enabled by default in Blender Ubuntu packages ( http://blender.stackexchange.com/questions/160/can-blender-export-to-the-collada-format ), and I don't even know if Collada (dae) import is supported. Perhaps it could make sense to convert the collision to stl and then edit the stl. Gazebo support both stl and dae , but stl is more then ok for collision purposes (dae has support for things such as colors or materials, but those are not required for collisions). Another road is to define the collision in SDF as a combination of primitive shapes instead of using a mesh (for example defining the hand as a box). 

* In the long term I think the right way of dealing with this issue is to properly define collision groups:  http://gazebosim.org/tutorials?tut=collide_bitmask&cat=physics . Unfortunately this collision groups are supported only on Gazebo6, that is not packaged for Ubuntu 12.04 . You have any tentative roadmap for migrating to a newer Ubuntu release or you are still blocked by the Eigen stuff?

traversaro, 2015-08-13:
Update: apparently the model self_collide is available only since Gazebo6 : https://bitbucket.org/osrf/gazebo/pull-requests/1480/implemented-self_collide-bool-for-model/diff .

rlober, 2015-08-13:
@traversaro 
Thanks for looking into this for me. This is all very good info. I am going to see what I can do about the collision meshes but for now it is not too critical.

Actually the real issue is our XDE simulator. We only have the binaries which were compiled for 12.04 so I wouldn't be able to use it at all if I move to 14.04. That being said I am working on it.

Although this issue isn't resolved per se, I think there is sufficient info for future users. 

Thanks for the help man!

closed 2015-08-13 by rlober