robotology-legacy/icub-gazebo-legacy #24: Remove ControlBoardWrapper warnings

traversaro, 2015-11-12:
See https://github.com/robotology/gazebo-yarp-plugins/issues/210#issuecomment-156236465 . 

@barbalberto @liesrock @EnricoMingo @arocchi 
Can you confirm me that the walkman superbuild is pointing to the `master` branch of YARP ? Thanks.

barbalberto, 2015-11-13:
Not exactly the master but updated at the beginning of month, commit 2fae6159e0d4f4a574c68188402d0857e07611d0. Anywayt it'll need to update again when we fix the newly found bug on networkClock #624

liesrock, 2015-11-13:
I confirm what @barbalberto wrote.

traversaro, 2015-11-13:
Thank you guys. :+1:

traversaro, 2016-12-07:
Fixed by https://github.com/robotology/icub-gazebo/issues/51 and https://github.com/robotology/icub-gazebo/issues/56 .

closed 2016-12-07 by traversaro