robotology-legacy/icub-gazebo-legacy #25: Error in the launch "Unable to read sdf file[.../icub.word]"

misaki43, 2015-12-10:
Hi,
 I just installed icub_gazebo with this link : "https://github.com/robotology-playground/icub-gazebo/issues" and I have some trouble when I want to launch "gazebo /[myPath]/icub-gazebo-master/world/icubword". 
--> first a warning (parser) says "converting a deprecated source[[myPath/icub-gazebo-master/icub/icub.sd".
--> second a warning says version 1.5 has been converted to 1.4
--> third an error says 'unable to convert 1.5 to 1.4
--> after, many errors linked to the reading element <dynamics> etc.
--> and after many errors about the reading of files.

Have you an idea of what is my problem ?

Sorry for my bad english...

traversaro, 2015-12-10:
Which version of Gazebo are you using?

misaki43, 2015-12-11:
It is the version 2.2.6.

traversaro, 2015-12-11:
I guess most of the model have been produced in this repo have been converted to URDF using a later version of Gazebo, and so the model is using the SDF version 1.5, while Gazebo 2.* uses version 1.4 . 

Sorry for not properly stating the incompatibility with Gazebo 2.* in the repo description, given that most of our users use the latest version of Gazebo we were not aware of this. 

Do you have a specific reason to use such an old version of Gazebo (perhaps compatibility with ROS Indigo?)

Providing models compatible with Gazebo 2.* may involve a bit of work (I don't currently have the clear idea on how to properly provide them, and if actually make sense). I wonder if in the time being it you could try to use a more recent version of Gazebo? Apparently it is possible (paying attention to ABI issue) to have later version of Gazebo even with ROS Indigo, see http://gazebosim.org/tutorials/?tut=ros_wrapper_versions#UsingaspecificGazeboversionwithROS . This may be worth even because there are several bugs in Gazebo 2.* that affect the iCub simulation, see https://github.com/robotology/gazebo-yarp-plugins#old-gazebo-versions .

misaki43, 2015-12-11:
Thanks for this information, I have no reason to use this 2.* version, in fact I choose it randomly.
So, you recommend the 1.9 version of this link: "http://gazebosim.org/download" or an other ? I don't find easily the 1.4's version.

traversaro, 2015-12-11:
The 1.5 / 1.4 versions mentioned in the previous comments are versions of the [`SDF` format](http://sdformat.org/spec). 
If you don't have any constraint, I would recommend (as it is reported in `gazebo-yarp-plugins` docs: https://github.com/robotology/gazebo-yarp-plugins#dependencies) to use the latest stable version of Gazebo, that is Gazebo 6.1 at the moment. Gazebo 6.1 fully supports SDF version 1.5 , so you should not have any problems with the models present in this repo.

misaki43, 2015-12-11:
Ok thanks ! I will try it next monday (I don't have my computer this week-end) and I will tell you if it works

traversaro, 2015-12-18:
Ok great! 
For the future: when you update an issue post a new comment, so participants on the issue will be notified of the update. If you edit an earlier comment, there is no way for the participants to the issue to know that you replied to the issue.

closed 2015-12-18 by traversaro