robotology-legacy/icub-gazebo-legacy #30: icub_with_hands gazebo model - Hands collision elements are not included

Yeshasvitvs, 2016-05-19:
@traversaro please send me the links for the shadow hand and other references

Thank you

traversaro, 2016-05-19:
A good reference for an highly articulated hand working fine in Gazebo is the MPL hand used in the Haptix simulator, see 
https://bitbucket.org/osrf/handsim/src/8fe03d4d113659c1cc04ea788792b1b7e995c267/models/mpl_haptix_left_forearm/model.sdf?at=default&fileviewer=file-view-default . 
Note that the substituted all the collision with cylinders, that is something that we may also consider. Highly detailed meshes are not ideal for simulation of contacts, and for example in the past we substituted the foot mesh with a box. 

Another good example is the Shadow Hand : https://github.com/shadow-robot/sr_common/blob/indigo-devel/sr_description/hand/ . This model is a bit more cumbersome to read, but you can check the value of the gazebo collision parameters in the `name.gazebo.xacro` files, for example https://github.com/shadow-robot/sr_common/blob/indigo-devel/sr_description/hand/xacro/thumb/thdistal.gazebo.xacro .

traversaro, 2016-05-19:
Hi @scpeters @j-rivero ,
  we are in the process of tuning the contact parameters of the iCub hand, and we were wondering if you have some good reference for existing humanoids hands modeled in Gazebo, from which we can get "inspiration" when doing contact parameter tuning. I already found the MPL hand used in Haptix and the Shadow Robot hand, but perhaps you know of other good examples. Thanks a lot in advance!

randaz81, 2016-05-19:
very important!
> Highly detailed meshes are not ideal for simulation of contacts, and for example in the past we substituted the foot mesh with a box.

j-rivero, 2016-05-19:
Hey Silvio:

> Hi @scpeters @j-rivero ,
> we are in the process of tuning the contact parameters of the iCub hand, and we were wondering if you have some good reference for existing humanoids hands modeled in [...]

@scpeters @hsu are probably the right people to ask for advice. From my poor experience, I can remember the good [Sandia Hand](https://bitbucket.org/osrf/sandia-hand/src/2532c504044ada285d24242d737801bc0cf69b18/ros/sandia_hand_description/?at=default) and the [iRobot hand](https://bitbucket.org/osrf/drcsim/src/583228ebd72677c36a393b3260f0c5f0bc7b0f87/irobot_hand_description/?at=default) from our times of DRC. Not sure if that is useful at all for you but I think that it is all I can mention.

scpeters, 2016-05-19:
The [MPL hand in handsim](https://bitbucket.org/osrf/handsim/src/8fe03d4d113659c1cc04ea788792b1b7e995c267/models/mpl_haptix_right_forearm/model.sdf?at=default&fileviewer=file-view-default) has had our most recent effort in parameter tuning. We also tend to focus our contact parameter tuning on ODE. The first thing I tend to do is to visualize contacts and try to pick a positive value of `min_depth` so that the contacts don't flicker on and off. We really should make a tutorial about this process.

traversaro, 2016-05-20:
Thanks @scpeters @j-rivero for the useful pointers.

frank-foerster, 2016-09-22:
Hi, in preparation for the (simulated) iCub grasping something I just noticed that its hand goes right through the object models. I assume it is this issue - missing hand collision elements - that is causing this, correct?  (as you can tell I'm quite new to the simulated version of the robot).
Has there been any progress on this since then?

Yeshasvitvs, 2016-09-22:
@frank-foerster yes it was missing the collision elements. I modified it for my work and you can use the model available here 
https://github.com/Yeshasvitvs/gazebo-yeshi/tree/master/icub_with_hands_cameras

frank-foerster, 2016-09-22:
@Yeshasvitvs I'll give it a try later today. Many thanks!

frank-foerster, 2016-09-23:
Hi @Yeshasvitvs ,
I just tried your `icub_with_hands_cameras` model but it seems that the iCub hands still penetrate the table surface 

(can't upload images from our lab, but here's link to the image)
[icub_with_hands_cameras penetrating table surface](https://goo.gl/photos/dQFT4WBZcMQbuAwS8)

I'll test it now with some other objects laying on the table surface.

frank-foerster, 2016-09-23:
I just tried the hand with the `Plastic Cup`, `Bowl`, and `Cardboard box` models from gazebo's online repository and the hand penetrates them all.

Yeshasvitvs, 2016-09-23:
@frank-foerster Just noticed. Regarding the hands the  reference is given to the icub model at https://github.com/robotology/icub-gazebo/tree/master/icub
So when I was working with it I added the collision elements same as the visual elements inside the meshes folder of my local repository. But the contact parameters have not been tuned yet and updated on the master. I suggest you to proceed this way for now

frank-foerster, 2016-09-23:
Ok, thanks for the info. As I have never added / tuned collision elements before I think I'll stick with the old simulator for the time being.

Yeshasvitvs, 2016-10-04:
@frank-foerster Hello, we just updated the collision elements for iCub hands. Please check and let us know if you have any troubles. Thank you

closed 2016-10-04 by Yeshasvitvs

frank-foerster, 2016-10-05:
Hi, yes, I've seen your update. Many thanks! I'm currently solving a different problem on the real robot, but will test the updated model thereafter and will let you now if I run into problems.
Thanks.