robotology-legacy/icub-gazebo-legacy #31: running iKinCartesianSolver with gazebo

tkelestemur, 2016-06-02:
Hello,

I am trying to use inverse kinematic solver with gazebo. I am able to use `iKinCartesianSolver` with iCub_SIM by running these commands in order:
1. `iCub_SIM`
2. `simCartesianControl --no_legs`
3. `iKinCartesianSolver --context simCartesianControl --part right_arm`
4. `./jointFollower` (my code to give position and orientation to end-effector [right_arm])

Now I am doing gazebo and adding iCub_with_hands 2, 3 and 4. step. Bu when I run `iKinCartesianSolver` I get this error:

```
[WARNING]Updating to newer yarp and the usage of controlBoardWrapper2 is suggested***
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
[INFO]cartesianSolver/right_arm: Checking if torso part is active ... not yet: still 0.6 [s] to timeout expiry
[ERROR]Device driver not available!
[INFO]cartesianSolver/right_arm closed
RFModule failed to open
```

How should I proceed ?

tkelestemur, 2016-06-03:
I figured out that to run iKinCartesianSolver, I need a context file for Gazebo simulation. When I run it for iCub_SIM I give `--context simCartesianControl` as the context file when I give it for gazebo it causes problems because of the port names.

traversaro, 2016-06-03:
Hi, please see this discussion here: https://github.com/robotology/QA/issues/115 .

tkelestemur, 2016-06-03:
hi @traversaro , I don't encounter with falling like mentioned in the discussion, instead, I see strange behavior of the robot. The controller code is [here](https://github.com/tkelestemur/icub_ros/blob/master/iCubSim/src/jointFollower.cpp).

Note : we should document the changes in cartesianControl.ini to make iKinCartesianController run.

traversaro, 2016-11-08:
@tkelestemur Sorry for the delay in a proper reply. 

Can you confirm that the problems that you see are similar to the one described by @frank-foerster in https://github.com/robotology/icub-gazebo/issues/37 ?

tkelestemur, 2016-11-14:
@traversaro I am sorry but I am not working on iCub anymore, so I am unable to confirm similarity.

traversaro, 2016-11-14:
No problem, I think anyhow this should be solved by https://github.com/robotology/icub-gazebo/issues/38 .

randaz81, 2016-11-14:
I'm closing this issue since:
- The original description refers to deprecated modules (i.e. simcartesiancontrol).
- There issue is duplicated. 
- Discussion refers to fixes which are outdated.

closed 2016-11-14 by randaz81