robotology-legacy/icub-gazebo-legacy #32: iCub Nancy URDF

UChervet, 2016-07-08:
Dear all,

We would like to simulate precisely our robot in gazebo and control it with WBI. The problem is that we need a custom URDF since our robot is peculiar : we have a V2 head, V1 arms and torso and V2 legs. If I'm not mistaken all the URDF for the other robots have V2 arms. Is there an URDF for a V1 robot ? Is it possible to combine parts from different URDF files to create our custom one or is there something else that we should know ? Could you help us making the correct URDF for iCubNancy01 ?

Thanks in advance.

@serena-ivaldi @bpapaspyros @jbmouret

traversaro, 2016-07-08:
Hi @UChervet , 
  most of the logic underlying the generation of iCub URDF models is maintained at https://github.com/robotology-playground/icub-model-generator (currently under revision). If it is ok for you, I would move this issue to that repo to address your question there.

traversaro, 2016-11-08:
To avoid duplication, I think we can discuss all the aspects of iCubNancy01 model generation at https://github.com/robotology-playground/icub-model-generator/issues/13 .

closed 2016-11-08 by traversaro