robotology-legacy/icub-gazebo-legacy #33: icub_with_hands incorrect pose due to left arm pose

Tobias-Fischer, 2016-07-11:
Hi,
I found that the pose of the `icub_with_hands` model is incorrect. The RPY is (-0.49, 0.93, 2.97) after inserting the model, rather than the expected (0.0, 0.0, 3.14) (as found e.g. in the normal `icub` models.

I quickly looked into it, and it seems like it is an issue with the left hand. When I comment out
```
    <!--plugin name='controlboard_left_arm_no_hand' filename='libgazebo_yarp_controlboard.so'>
      <yarpConfigurationFile>model://icub/conf/gazebo_icub_left_arm_no_hand.ini</yarpConfigurationFile>
      <initialConfiguration>-0.52 0.52 0 0.785 0 0 0.698</initialConfiguration>
    </plugin--> 
```
in the SDF file, the behavior is as expected when inserting the model (afterwards drifting a little due to the different masses on the left and right side of the arms in that case).

It would be great if someone could look into this. I guess it's just a wrong sign somewhere, but I have no idea where to start looking for it.

Thanks, Tobi
/cc @randaz81 @traversaro

traversaro, 2016-11-08:
Hi @Tobias-Fischer ,
  sorry for the late response. 
What do you mean by `pose` of the model?
The one that you see in the side panel? 

[image]

Tobias-Fischer, 2016-11-08:
Hi @traversaro,
it's been a while that I haven't been working with this. Yes, I do mean the pose in the side panel, and as far as I can remember visually you can see that the hand is in a weird configuration as well.
If you can't reproduce on your system, I'll give it a try and see whether it still happens here.

Thanks, Tobi

Tobias-Fischer, 2016-11-08:
Maybe related to https://github.com/robotology/icub-gazebo/issues/35 ?

traversaro, 2016-11-08:
Ok, so it is the `pose` of the `l_hand`, not of the overall robot model, I think I got it and you are right, it is probably related to https://github.com/robotology/icub-gazebo/issues/35 .

Tobias-Fischer, 2016-11-08:
Yes, the pose of `l_hand` :).

randaz81, 2016-11-08:
Not sure if I have understood correctly, also because I cannot reproduce issue #35.
I'm thus reinterpreting this issue in following way: "The parking position of the wrist when I start the robot in the simulator, is not 0 as in the real robot".
I fixed this with commit b29cc396891af548a23e6063cb0a3ecac6122d68

Tobias-Fischer, 2016-11-08:
Hi @randaz81 and @traversaro,
I just had a quick look, and I was referring to the pose of the `iCub` rather than the pose of the `l_hand`. See the attached screenshot. The orientation should be `(0.0, 0.0, 3.14)` but is actually something else. See also the tilted box within Gazebo.

Best,
Tobi

[image]

randaz81, 2016-11-08:
Ok, I see. I'm checking..

randaz81, 2016-11-09:
There's something tricky here. Numbers put in the model seems more or less right to me.

However I noticed the following facts:
- If I swap the order in which the two includes tags are written inside the .sdf, the result is wrong in a different way. If I remove the two includes, the model pose is ok. (1)
- If I remove the initialConfiguration tags, or if I put an initial configuration equal to zero, the model pose is ok. Swapping the order of motioncontrol plugins in the sdf file does not have any effect. 
- The issues is not due to any of the following: the visual meshes, the collision meshes, the masses or the inertias of the robot hands. I also removed all the fingers links and joints from the two included models, keeping only the base link. I put (0 0 0 0 0 0) as default pose for this base link. The issue is still present.
- I currently have not found any information about how gazebo computes the model bounding box and the initial pose.

MEMO:
Because of (1), I'm currently thinking to a bug in the gazebo engine. 
I'd like to verify this hypothesis with a workaround in the GazeboYarpControlBoardDriver::gazebo_init() in order to postpone the execution of m_jointPointers[i]->SetPosition(0,initial_config[i]) A possibility is to move it in the onUpdate() function. An additional confirmation could be given by preparing a single plugin configuration which sets the initialConfiguration simultaneously on the left and right arm.

Tobias-Fischer, 2016-11-09:
Thanks for looking into this @randaz81! Let me know if I can help.

randaz81, 2016-11-09:
Commit 9e13fb5c66 almost fixes the issue (A).
The issue was fixed by moving down the include tag (nice, isn't it??)
I still don't know why the order of tags (1) `and the values contained in intialConfiguration (2) affect the robot pose. Most likely there's a bug in gazebo xml parser plus a strange interaction between the gazebo engine and our motion control plugin... (but I currently do not have time to open an issue on gazebo bug tracker...)

PS:
(A) The exact pose orientation is not (0 0 3.14) but (0.0009 -0.011 3.140029), I don't know why.

closed 2016-11-18 by randaz81