robotology-legacy/icub-gazebo-legacy #34: (In)compatibility of ARE with icub_with_hands model?

frank-foerster, 2016-09-20:
Dear all, 

I'm new to gazebo but have been playing around with it for a while now to see whether I can adopt it as the new standard iCub simulator (as recommended by @kt10aan). 
I'm in particular interested in using the actionsRenderingEngine in order to generate grasping and similar hand-related behaviours, so I tried to use the icub_with_hands model. 
When loading this model I noticed that hand-related port names generated by the gazebo model differ from the ones used on the real robot as well as the 'old' simulator.
On the old simulator the hands are reachable via the ports
`/icubSim/right_arm/..`
`icubSim/left_arm/..`

whereas on the icub_with_hands gazebo model I see ports such as

`/icubGazeboSim/left_hand_thumb/..`
`/icubGazeboSim/left_hand_index/..`

etc. 

As I'm also new to the ARE, and as its documentation is silent on which precise port names it assumes, I was trawling through its source code trying to figure out whether it can be configured to use these finger-based port names / port prefixes as created by the gazebo model. But as far as I understand the code this doesn't seem to be the case. 

For example in `../icub-main/modules/actionsRenderingEngine/src/MotorThread.cpp`, function
`bool MotorThread::threadInit()`, I can see that the port names appear to be hardcoded to fit the "port naming model" of the real robot / old simulator, as for example in the line

`optLeftArm.put("remote", ("/"+robot+"/left_arm").c_str()`

So am I correct that the gazebo icub_with_hands model is incompatible with ARE or is there a way to adapt either the model or ARE to make the two work with each other? 

Many thanks

randaz81, 2016-09-20:
Hi,
The correct names are the ones used by ARE. 
The gazebo simulator should open THOSE ports (left_arm ecc) PLUS additional ports, which are the ones that you are mentioning (left_hand_thumb ecc) 
If you see only the additional ports, there's a problem in the simulator model. I'll check the model.

frank-foerster, 2016-09-20:
Just ran the model again and you're right. The /.._arm ports are created. I probably didn't see those before because I had some other issues with my local installation. 
I'll have a go with ARE now (if I get to run it properly) and will close this issue later, if I don't encounter further problems. 
Thanks for the hint.

frank-foerster, 2016-10-25:
After getting all the other modules which are typically started for the old simulator to run with Gazebo I just reached the point at which ARE itself is quitting on me due to the Gazebo simulator not creating the expected skin-related ports: 

`[ERROR]*** actionsRenderingEngine/tactile: unable to connect to /icubGazeboSim/skin/left_hand_comp
[INFO]*** actionsRenderingEngine/tactile: closing ...`

So, my question: Is skin already available on Gazebo iCub, and if yes, how do I activate it?

And if not, is there a way to run ARE or the underlying action primitives if one wants to grasp without skin?

frank-foerster, 2016-10-25:
Thanks for the reply. Once I know a bit more what I'm doing and finished an urgent experiment, I might be able to help.

pattacini, 2016-10-25:
Hi @frank-foerster 

[`ARE`](http://wiki.icub.org/iCub/main/dox/html/group__actionsRenderingEngine.html) wraps around [`actionPrimitives`](http://wiki.icub.org/iCub/main/dox/html/group__ActionPrimitives.html) to perform action sequencing, which in turn makes use of [`perceptiveModels`](http://wiki.icub.org/iCub/main/dox/html/group__PerceptiveModels.html) to deal with the required perception underlying the grasp.

As of now, two [**implementations**](http://wiki.icub.org/iCub/main/dox/html/group__percmod__Implementations.html) are available from within `perceptiveModels`:
- [`tactileFingers`](http://wiki.icub.org/iCub/main/dox/html/group__TactileFingers.html)
- [`springyFingers`](http://wiki.icub.org/iCub/main/dox/html/group__SpringyFingers.html)

You might want to test `springyFingers` within Gazebo then.

Further resources on this: https://github.com/robotology/QA/issues/34

frank-foerster, 2016-10-25:
Hi @pattacini ,
thanks for the hint to the springyFingers implementation, that was the kind of hint I was looking for.

closed 2016-10-25 by frank-foerster