robotology-legacy/icub-gazebo-legacy #35: icub_with_hands gazebo model joint6(wrist_yaw) position cannot be moved using yarpmotorgui

Yeshasvitvs, 2016-09-29:
Hello, 

In the icub_with_hands gazebo model joint6(wrist_yaw) position cannot be moved using yarpmotorgui. The initial position is at one of the joint limit values of 24.57 and cannot be moved towards the other joint limit side. This problem does not exists with icub with no hands model 

Thank you

randaz81, 2016-11-08:
I'm sorry but I cannot reproduce this problem. 
I'm using icub_with_hands model, and I'm trying to move joint 6 (wirst_yaw) on both arms from its initial position. I can easily move it from 25 to -19. Maybe I have not understood correctly, or this issue has been already fixed in the past. Can you check it?

randaz81, 2016-11-10:
The fix robotology/QA#155 should have improved joint effort limits. I think that this problem on the wrist_yaw should not happen anymore...

Yeshasvitvs, 2016-11-10:
@randaz81 ciao marco, 

I just tried it and the problem doesn't exist anymore. Thanks for the fix. I also noticed a lot twitching in the fingers even when I am not moving the fingers. I am not sure if it just on my machine, please let me know

randaz81, 2016-11-10:
Uhm, small inertias may lead to numerical instability, however I haven't spotted this issue in my tests.
Let's close this issue, and discuss this new problem on a separated thread.

closed 2016-11-10 by randaz81