robotology-legacy/icub-gazebo-legacy #37: unstable iCub behaviour when running ARE in Gazebo

frank-foerster, 2016-10-31:
After having solved the issues with running ARE in conjunction with the Gazebo iCub (see #34 ),
I tested various functions from ARE with the new simulator which resulted in rather unstable robot behaviour resulting from nearly all functions except for the `home head` and `home hands` commands. 

(I'm using the springyFingers model as the skin is not implemented on Gazebo yet).

With `unstable behaviour' I mean that after executing the respective command - some commands are also executed in a urrecognizable/unsuccessful  manner - the robot seems to become very unstable with its upper body sometimes falling over onto the table or its upper body hanging precariously over to one side.

I'm not sure if this is related to issue  #31 as ARE obviously relies on the solvers to execute the commands.

I used a 'fixed' model where the iCub's legs are fixed, so it could not possibly fall over completely. 
As the above descriptions are rather vague I took some screen recordings which show these `instabilities': 

I did not observe any significant error or warning messages coming from the ARE module itself except in the case when executing the `calibration table` command (see below). I executed the other relevant modules via the yarpmanager so I didn't see their output. 

**home** commands: `home head` and `home hands` appear to execute ok, but after `home all`
the robot becomes unstable
Video: executing [home commands](https://vimeo.com/189315359)

**calibration table** command: this is the only command where I got a command line message indicating that something went wrong, namely, an `[INFO]` message saying that a timeout expired while reaching was executed followed by a `[WARNING]` saying that the table height was not found despite the robot clearly pressing onto the table surface. 
I compared this with the output of the `old' simulator and there the `[WARNING]` says that the table height was found. So somehow the exertion of pressure is onto the hands is not detected.
Video: executing [calibration table](https://vimeo.com/189315519)

**expect** command: here the execution look considerably different compared to when executing the command on the old simulator.
Video: executing [expect](/https://vimeo.com/189316214)

**exploration torso** command: this one executed in a hardly recognizable manner and goes bad fast.
Video: executing [exploration torso] (https://vimeo.com/189316356)

cc  @(pattacini)

traversaro, 2016-11-08:
Thanks a lot for the feedback, @frank-foerster ! 

I think most of the issue that you are describing are due to poorly-tuned low-level PID parameters, see https://github.com/robotology/icub-gazebo/issues/38 . 

One additional question: which table model did you for testing the "calibration table" command?

frank-foerster, 2016-11-08:
Hi Silvio (@traversaro), thanks for the hint to the PID paramters. I don't know how to tune these though and have resorted to using the old simulator for the time being.
The table I used was the standard [table model](http://models.gazebosim.org/table/) from the Gazebo repository, where I adapted the height: I set z to 0.33m (instead of 0.5m) in the model inspector for all four legs under both the 'Visual' as well as the 'Collision' tab.

randaz81, 2016-11-08:
> the robot seems to become very unstable with its upper body sometimes falling over onto the table 

Regarding the stability, check also https://github.com/robotology/QA/issues/155 which has been hopefully just fixed.

One caveat: every time a model is updated (like in this case with new dynamic parameters), optimal PID gains might change and the controller stability should be re-checked.

frank-foerster, 2016-11-09:
As far as I can tell the recent update didn't fix the issues. I updated my local copy of icub-gazebo and used the regular icub_with_hands model this time after having used a modified one last time. 
I also lowered the table so it roughly matches the table height of the old simulator - it's still a little bit lower though I think.
I just tried the `home all` and `calibrate table` commands, but in both cases the robot falls over after a while. 
Video of [home all](https://vimeo.com/190897383)
First video of [calibrate table](https://vimeo.com/190897503) - table too far away?
2nd video of [calibrate table](https://vimeo.com/190897622) with the table being closer to the iCub.

A fixed model might help with the falling over, but in case of `home all` it still seems to become unstable before falling down.

pattacini, 2016-11-09:
Working nicely, thanks a lot!

pattacini, 2016-11-09:
Sorry I commented on the wrong issue 😞

frank-foerster, 2016-11-10:
Just to complete the above test: I re-tested the new model using the 'icub_with_hands_fixed' "wrapper", 
and the outcome was the same. As the robot is fixed, it doesn't fall over any more - instead it "hang over":

Video of [home all command](https://vimeo.com/190994843) with fixed model

randaz81, 2016-11-10:
Did you used the very last version (updated one approx 1-2 hours ago) ?

randaz81, 2016-11-10:
Ps: Would you mind to send me your scripts/configuration files to run are + the cartesian controllers on gazebo? Thank you

frank-foerster, 2016-11-11:
As it turns out I haven't hat the very latest version, but the one before that. Just ran another `git pull` and will test now again. I'll report on the result here and will send you my scripts + config files after that. Thanks

frank-foerster, 2016-11-11:
I just re-tested the `icub_with_hands` model with the most recent updates but still get some unstable behaviour. Just sent you the script and config-files I am using to run run ARE in Gazebo.

randaz81, 2016-11-14:
Hi @frank-foerster, we found multiple issues, at different levels (including yarp_dev), preventing both ARE and the cartesian controller to properly work with Gazebo. I'll notice you as soon as we push the fixes. 
Additionally, I checked your scripts, and I noticed that you are using a module simCartesianControl, which has been deprecated and removed from icub-main (See https://github.com/robotology/QA/issues/146). I'll send you updated scripts. 

PS: To improved the readability of this topic, I'll soon remove the last few messages.

frank-foerster, 2016-11-15:
Hi,
I'm aware that simCartesianControl is outdated, I just didn't get around to update as of yet. 
But I'll update soon. Thanks for looking into this.

traversaro, 2022-01-13:
I think we can close this as it is a duplicate of https://github.com/robotology/icub-models/issues/120 .

closed 2022-01-13 by traversaro