robotology-legacy/icub-gazebo-legacy #38: Tune low-level gains of joint level controllers

traversaro, 2016-11-08:
Several users are experiencing issues that (I think) are related to not properly tuned joint-level PID position loops of the iCub model in Gazebo, see:
* https://github.com/robotology/QA/issues/155
* https://github.com/robotology/icub-gazebo/issues/37
* https://github.com/robotology/icub-gazebo/issues/35

We should tune the position gains with respect to some clearly defined scenario, such as the ARE one described by @frank-foerster in https://github.com/robotology/icub-gazebo/issues/37 . For reference we should document on which scenario we tuned the gains, such that in the future we can simplify the process of re-tuning the gains for different robots (such as R1).

traversaro, 2016-11-14:
Renamed the issue after the recent findings.

randaz81, 2016-11-14:
There a bunch of bugs that prevent ARE to work with gazebo. I'll remove this message as soon the fixes are be pushed.

traversaro, 2022-01-13:
Is this still the case @pattacini ? In any case, I think the proper place for this issue is icub-models.