robotology-legacy/icub-gazebo-legacy #39: Cleanup models in the repo

traversaro, 2016-11-08:
Only models of general relevance for should be committed in this repo, not random stuff.

traversaro, 2016-11-10:
Some people in the [DIC](https://www.iit.it/lines/dynamic-interaction-control) Lab at IIT (@DanielePucci, @gabrielenava) have several experiment specific models in this repo that I think should be moved to a specific repository, I will temporarily call it `icub-dic-gazebo`. 

For all the other models I propose :

| Model |  Action | Comments |
|:------:|:-------:|:-----------:|
| `ContactTest` | Move to `icub-dic-gazebo` | |
| `conf` | ?? | I don't have a clear idea, it contains an outdated configuration file for yarprobotmotorgui for the icub without hands model. It should be probably just removed. |
| `double_pendulum` | ?? | Move it to `gazebo-yarp-plugins` examples or leave it here? |
| `iCubHeidelberg01_fixed` | keep | | 
| `iCubHeidelberg0` | keep | Keep, but move to the model generated from `icub-model-generator` and possibly change the name for consistency. |
| `icub` | keep, rename to `icub_without_hands` if it does not break to many models. | A lot of people doing balancing/walking stuff depend on that, and at the moment the model with camera/hands is quite more computationally extensive, so we need to keep it. Note that a rename would mean that we need to change all the models that refer to `model:///icub/<...>` | 
| `icub_fixed` | Move it to the iCub with hands and cameras | I think it would make more sense to have a full iCub (with cameras and hands) in the fixed version. | 
| `icub_no_ft` | remove | It was necessary when Gazebo did not support the fixed joints. | 
| `icub_on_seesaw` | Move to `icub-dic-gazebo` | CoDyCo/DIC specific stuff | 
| `icub_with_hands` | Merge with `icub_with_cameras` | See https://github.com/robotology/icub-gazebo/pull/41 | 
| `icub_with_cameras` | Remove by merging with `icub_with_hands` | See https://github.com/robotology/icub-gazebo/pull/41 | 
| `inclined_plane` | Move to `icub-dic-gazebo` |  |
| `old_and_deprecated/icub_with_hands' | Remove | | 
| `seesaw` | Move to `icub-dic-gazebo` | | 
| `world` | Rename to worlds | Add a model with a table, to have an environment similar to `iCubSim` |

To simplify integration of this (and other gazebo models repo) in superbuild, I propose to add an installation step to this repos, that install models to `<prefix>/share/gazebo/models` and worlds to `<prefix>/share/worlds`, mimicking gazebo installation layout and simplifying the process of adding all the models of a superbuild to Gazebo (we just need to add `<prefix>/share/gazebo/models` to GAZEBO_MODEL_PATH, not every single repo separetly.

frank-foerster, 2016-11-11:
I used the `icub_with_cameras` until yesterday and made my own `-fixed` version of that, I had to switch back to the `icub_with_hands` model in order to test the latest changes. So I'd welcome a cleanup and a merger of the camera functionality into the "standard" `_with_hands` model, and also, having a full iCub in the `_fixed` model. I haven't used superbuild so far, so I can't say anything about that.

traversaro, 2016-11-11:
@frank-foerster  Thanks for the feedback. I plan to work on this next week.

traversaro, 2016-11-17:
I tested https://github.com/robotology/icub-gazebo/issues/48 , and it is working fine. Given that most of the users of the models moved to https://github.com/robotology-playground/icub-gazebo-wholebody/commit/6be6ad8ff88e0368ab248bae4fbc60d91a0ae5d6 are now in Mexico, I think I can merge, and users of those models can simply catch up when they return.

cc @DanielePucci @gabrielenava @francesco-romano @S-Dafarra

traversaro, 2016-11-17:
Fixed by https://github.com/robotology/icub-gazebo/pull/48 .

closed 2016-11-17 by traversaro