robotology-legacy/icub-gazebo-legacy #40: iCub with hands and Cameras

Yeshasvitvs, 2016-11-10:
@traversaro @randaz81 
Ciao Guys, 

I see that we dont have any gazebo model with hands and cameras together. I am not sure if there is any particular reason for it. If not can I push it to master? Please let me know 

Thank you

traversaro, 2016-11-10:
I think it make sense to just add cameras to `icub_with_hands` and then to remote `icub_with_cameras`.

traversaro, 2016-11-10:
@Yeshasvitvs In general for proposing any modification feel free to open a PR so we can discuss directly there.

Yeshasvitvs, 2016-11-10:
Yes that is how I am using it. Added the camera plugins and conf files and modified the sdf of hands model 

Cheers

claudiofantacci, 2016-11-10:
I am also interested in a Gazebo iCub model with hands (hence torso + arms) and cameras.

@Yeshasvitvs do you have a model which is also fixed, say, at the ROOT so that it does not fall?

Yeshasvitvs, 2016-11-10:
@claudiofantacci  I have a floating base model with hands and cameras

claudiofantacci, 2016-11-10:
Yeah that fine, will do the job 😄 !
Thanks!

closed 2016-11-10 by Yeshasvitvs

Yeshasvitvs, 2016-11-18:
see https://github.com/robotology/icub-gazebo/pull/47