robotology-legacy/icub-gazebo-legacy #44: deprecated PID format detected

randaz81, 2016-11-14:
PID gains written in the ini files should be expressed using the following format:
```
[POSITION_CONTROL]
controlUnits  metric_units
controlLaw    joint_pid_gazebo_v1
kp           20.0  20.0
kd           0.0   0.0
ki           0.01   0.01
maxInt       9999  9999
maxOutput    9999  9999
shift        0     0
ko           0     0
stictionUp   0     0
stictionDwn  0     0
```

and not as:
```
[GAZEBO_PIDS]
#Left Arm
Pid0 100.0 10.0 10.0 9999 9999 9 9
Pid1 100.0 10.0 10.0 9999 9999 9 9
Pid2 100.0 10.0 10.0 9999 9999 9 9
Pid3 100.0 10.0 10.0 9999 9999 9 9
Pid4 100.0 10.0 10.0 9999 9999 9 9
Pid5 100.0 10.0 10.0 9999 9999 9 9
Pid6 100.0 10.0 10.0 9999 9999 9 9
```

The latter also using a rad2deg conversion factor that make the robot configuration inconsistent with the values returned by yarp interfaces.

traversaro, 2016-11-17:
Relevant gyp issues: 
* https://github.com/robotology/gazebo-yarp-plugins/issues/119
* https://github.com/robotology/yarp/issues/491
* https://github.com/robotology/gazebo-yarp-plugins/issues/218

traversaro, 2016-11-17:
cc @Yeshasvitvs

traversaro, 2016-12-07:
Fixed by https://github.com/robotology/icub-gazebo/issues/51 .

closed 2016-12-07 by traversaro