robotology-legacy/icub-gazebo-legacy #54: Add l_sole to SDF

rlober, 2016-11-29:
Hey guys,

I was wondering if you could add the `l_sole` and `r_sole` as links in the main iCub model sdf? We are using the soles for odometry and in order to visualize frames in gazebo properly I need to grab the initial `l_sole` pose from gazebo to get the rototranslation from gazebo to the robot, that way, all the frames I have expressed in the robot world can be expressed in the gazebo world.

Thanks in advance.

traversaro, 2016-11-30:
I have a long answer to this question, but I do not have time to write it at the moment. :D 
(Hint: it is related to https://github.com/robotology/codyco-modules/issues/39 , http://discourse.ros.org/t/urdf-ng-link-and-frame-concepts/56 ). 

In a nutshell: if you have a configuration that is working for you (i.e. the robot is not exploding even if this `l_sole` and `r_sole` ``link'' have a negligible mass and software-wise is working for you) can you open a PR? Then we will slowly try to add this to the model generator, see https://github.com/robotology-playground/icub-model-generator/issues/25 .

rlober, 2016-11-30:
Wow, this was a much bigger can of worms than I could have ever imagined! I think I will just let this little feature request go for the time being.

closed 2016-11-30 by rlober