robotology-legacy/icub-gazebo-legacy #61: simulated icub on gazebo drifts when using Gaze controller

pattacini, 2017-07-11:
@valeriosperati commented on [Mon Jul 10 2017](https://github.com/robotology/QA/issues/242)

Hi, I have a code that initialises the gaze controller on simulated icub and, on a keypress event, directs the gaze on a fixed point using lookAtMonoPixel() function.
It works as expected with icub_SIM but fails using gazebo: the head robot immediately drifts after the initialisation of gaze controller

[video example](https://drive.google.com/open?id=0B7eFeb2P2Ie7bUV6NVFGNUlrb00)

 ```c++
    PolyDriver m_clientGaze;
    Property option_gaze;
    option_gaze.put("device","gazecontrollerclient");
    option_gaze.put("remote","/iKinGazeCtrl");
    option_gaze.put("local","/client/gaze");
    
    if(!m_clientGaze.open(option_gaze))
    {
	cout<<"\n[FATAL], problems in opening gaze controller...\n\t";
	cout<<"Is iKinGazeController running?...\n\n";
        std::string exit(0);
    }
    
    if(m_clientGaze.isValid())
    {
	m_clientGaze.view(iGaze);
    }else
    {
	cout<<"\n[FATAL], problems in viewing gaze controller...\n";
        std::string exit(0);
    }
    
    iGaze->setNeckTrajTime(0.8);
    iGaze->setEyesTrajTime(0.4);
    
    // put the gaze in tracking mode, so that
    // when the torso moves, the gaze controller 
    // will compensate for it
    iGaze->setTrackingMode(true);
    iGaze->setStabilizationMode(true);
```

iKinGazeCtrl is launched with parameters:

`iKinGazeCtrl --robot icubSim --from configSim.ini`

Gazebo is launched  with parameters:

`gazebo setupV0.sdf`

where setupV0.sdf is my starting configuration running  icub_with_hands and the worldInterface plugin:

```xml
<?xml version='1.0'?>
<sdf version='1.4'>
  <world name="my_world">

    <!-- Light -->
    <include>
      <uri>model://sun</uri>
    </include>

    <!-- Ground Plane -->
    <include>
      <uri>model://ground_plane</uri>
    </include>

   <!-- icub  and worldinterface plugin -->
   <model name="iCub_with_hands">
   	<include>
	    <uri>model://icub_with_hands</uri>
    	</include>
        <plugin name="world_interface" filename="libgazebo_yarp_worldinterface.so">
           <yarpConfigurationFile>model://worldInterface/conf/worldInterface.ini</yarpConfigurationFile>
        </plugin>
   </model>
</world>
</sdf>
```

The standard output of gazebo is the following:

[output.txt](https://github.com/robotology/QA/files/1135701/output.txt)

Thanks for help,
Valerio

---

@pattacini commented on [Mon Jul 10 2017](https://github.com/robotology/QA/issues/242#issuecomment-314145587)

@valeriosperati I'm afraid that what you've seen is very much related to a number of open issues on https://github.com/robotology/icub-gazebo/issues that stem from the fact that `icub-gazebo` still need quite a lot of tuning of the low-level PID control layer.

See for example https://github.com/robotology/icub-gazebo/issues/38.

Therefore, if you don't mind, I'd rather move this issue therein. 

---

@valeriosperati commented on [Tue Jul 11 2017](https://github.com/robotology/QA/issues/242#issuecomment-314392928)

Yea, sure!

emilio-cartoni, 2017-07-25:
I had the same issue of the neck "drifting" when launching the tutorial_gaze_interface

Neck behavior was fixed by enabling position_direct mode in iKinGazeCtrl following (the reverse of) these instructions: https://github.com/robotology/icub-main/issues/129#issue-64464506
According to this (https://github.com/robotology/gazebo-yarp-plugins/issues/240) velocity mode does not work on Gazebo, while position_direct (used for the neck) and mixed mode (used for the eyes) should work.

However, the tutorial_gaze_interface still seems to behave differently on Gazebo compared to iCub_SIM, since, for example, the robot does not seem to wait up until torso has completed its movements before initiating another circle of fixations.

pattacini, 2017-10-16:
Hi @valeriosperati @emilio-cartoni 

In view of #62, try out with the contexts provided [here][1]. I made the gaze controller run within Gazebo using the `position-direct` mode.

[1]: https://github.com/robotology/icub-gazebo/issues/62#issuecomment-336461717

pattacini, 2017-10-19:
Sorry @emilio-cartoni 
Didn't realize you already figured it out.

pattacini, 2017-10-21:
Hi @emilio-cartoni 

https://github.com/robotology/icub-tutorials/commit/e019bc80c75d30c6b202aeffc67498a62914f0f1 should have fixed the different behaviors you saw on iCub_SIM and Gazebo.

I think we can close this.

closed 2017-10-21 by pattacini