robotology-legacy/icub-gazebo-legacy #62: ARE "unable to connect to the server rpc port" error

dcam0050, 2017-10-11:
I'm having an issue with `ARE` while using `icub with hands` model.

It starts off OK but then fails with ```[ERROR] unable to connect to the server rpc port!```

I've attached here the log for ARE launched using --configSim.ini. 

[ARE_icubgazebo_log.txt](https://github.com/robotology/icub-gazebo/files/1376660/ARE_icubgazebo_log.txt)

These are the modules that I'm running:

iKinCartesianSolver --context simCartesianControl --part right_arm
iKinCartesianSolver --context simCartesianControl --part left_arm
iKinGazeCtrl --from configSim.ini
wholeBodyDynamics --autoconnect --headV2 --dummy_ft
gravityCompensator --headV2
actionsRenderingEngine --from configSim.ini

All of them work well with the exception of actionsRenderingEngine.

dcam0050, 2017-10-12:
I figured out the issue which was the absence of  yarprobotinterface --context simCartesianControl

Now `ARE` starts OK however when I issue a `home_all` the icub falls forward

[image]

pattacini, 2017-10-12:
Hi @dcam0050 

Gazebo is a bad company to spend the night with unless you're now on the other side of the Earth 😉

Concerning the stability of ARE within Gazebo there're already a number of quite longstanding issues (see #37 #38) I would rather point you at.

As a workaround, you might want to give ARE a try in stiff mode. To do so, disable the parameter [`impedance`][1] within the ARE config file.

[1]: https://github.com/robotology/icub-main/blob/master/app/actionsRenderingEngine/conf/config.ini#L46

dcam0050, 2017-10-12:
Pretty bad company indeed during the night :sleepy: 

I did try setting impedance off because when impedance is on, there are warning messages issued in the gazebo terminal when a `home_all` is issued ```[WARNING]Compliant velocity control not yet implemented``` so I tried impedance off however the result was the same.

Could it be an issue of configuration files? Under robots directory in .local/share/yarp I have iCubSheffield is it a case that I need a gazebo specific configuration in the robots directory?

pattacini, 2017-10-13:
@dcam0050 I've been playing around with `ARE` in Gazebo today and, eventually, I made it run successfully, at least the `home all` command.

First, yes, please get rid of the `YARP_ROBOT_NAME` env variable for these tests, at least temporarily, to avoid any possible interaction while yarp retrieves contexts.

The key point is to configure the Cartesian controller as well as the gaze controller to run in `position-direct` mode (as it happens on the real robots) since the `velocity` mode used along with `iCub_SIM` doesn't seem to be well handled.

Finally, we need to make `ARE` use springy models for the control of the fingers rather than the tactile models: the skin is not (fully) implemented yet, as far as I know.

To this end, you might want to re-use the [contexts][1] I've come up with during my tests. Just unzip the archive in your `~/.local/share` path (beware of possible overwriting).

The archive does also contain the application I used to launch the involved modules.

Once everything is set up, you can test the following:
```sh
$ yarp rpc /actionsRenderingEngine/cmd:io
>> home all
```

[1]: https://github.com/robotology/icub-gazebo/files/1382802/yarp.zip

pattacini, 2017-10-13:
After these investigations, it might be worth looking back to #37 and #38.

@traversaro @randaz81

pattacini, 2017-10-19:
@dcam0050 any news on this?

pattacini, 2017-10-23:
Closing for now.

closed 2017-10-23 by pattacini

dcam0050, 2017-10-25:
Hi @pattacini sorry for the delay in trying this out. I just tried it and I confirm it works great!

pattacini, 2017-10-25:
Cool!
Thanks for reporting @dcam0050

robertorovella91, 2019-01-21:
Hello everyone, 
we are trying to transfer our simulation on the gazebo but I'm having problems with the gaze controller. Infact, while for the movement of the limbs and the neck it is possible to activate respectively the position control and the position-direct from the configuration file, for the eyes it seems that they can only use the mixed control that causes the drift of the eyes. Is it possible to set the position control also for the eyes?
Kind regard,
Roberto Rovella

traversaro, 2019-01-21:
Hi @robertorovella91 , 
   I suggest to open a new issue (for example in https://github.com/robotology/QA) instead of commenting old closes issues, this will give you more visibility to your question. Thanks a lot!

pattacini, 2019-01-21:
@robertorovella91 
Unfortunately, I don't think there is any good news in this respect.
@xenvre might add insights.

robertorovella91, 2019-01-21:
So there is not possible to run gaze controller on gazebo ?