robotology-legacy/icub-gazebo-legacy #63: Ports prefix conflict while pawning multiple composite models

Yeshasvitvs, 2018-01-30:
As mentioned in this [issue](https://github.com/robotology/gazebo-yarp-plugins/issues/93) while spawning multiple icub robots into gazebo `gazeboYarpPluginsRobotName icubSim` inside `gazebo_icub_robotname.ini` is commented out. 

All the FT sensor ports and inertial ports open correctly while spawning two base icub models inside gazebo

```
registration name /iCub/left_arm/analog:o ip 10.255.54.112 port 10008 type tcp
registration name /iCub/left_arm/analog:o/rpc:i ip 10.255.54.112 port 10007 type tcp
registration name /iCub/left_foot/analog:o ip 10.255.54.112 port 10191 type tcp
registration name /iCub/left_foot/analog:o/rpc:i ip 10.255.54.112 port 10190 type tcp
registration name /iCub/left_leg/analog:o ip 10.255.54.112 port 10189 type tcp
registration name /iCub/left_leg/analog:o/rpc:i ip 10.255.54.112 port 10188 type tcp
registration name /iCub/right_arm/analog:o ip 10.255.54.112 port 10010 type tcp
registration name /iCub/right_arm/analog:o/rpc:i ip 10.255.54.112 port 10009 type tcp
registration name /iCub/right_foot/analog:o ip 10.255.54.112 port 10195 type tcp
registration name /iCub/right_foot/analog:o/rpc:i ip 10.255.54.112 port 10194 type tcp
registration name /iCub/right_leg/analog:o ip 10.255.54.112 port 10193 type tcp
registration name /iCub/right_leg/analog:o/rpc:i ip 10.255.54.112 port 10192 type tcp
registration name /iCub_0/left_arm/analog:o ip 10.255.54.112 port 10230 type tcp
registration name /iCub_0/left_arm/analog:o/rpc:i ip 10.255.54.112 port 10229 type tcp
registration name /iCub_0/left_foot/analog:o ip 10.255.54.112 port 10224 type tcp
registration name /iCub_0/left_foot/analog:o/rpc:i ip 10.255.54.112 port 10223 type tcp
registration name /iCub_0/left_leg/analog:o ip 10.255.54.112 port 10222 type tcp
registration name /iCub_0/left_leg/analog:o/rpc:i ip 10.255.54.112 port 10221 type tcp
registration name /iCub_0/right_arm/analog:o ip 10.255.54.112 port 10232 type tcp
registration name /iCub_0/right_arm/analog:o/rpc:i ip 10.255.54.112 port 10231 type tcp
registration name /iCub_0/right_foot/analog:o ip 10.255.54.112 port 10228 type tcp
registration name /iCub_0/right_foot/analog:o/rpc:i ip 10.255.54.112 port 10227 type tcp
registration name /iCub_0/right_leg/analog:o ip 10.255.54.112 port 10226 type tcp
registration name /iCub_0/right_leg/analog:o/rpc:i ip 10.255.54.112 port 10225 type tcp

```

But when `iCub (fixed)` model, which uses composite `icub_with_hands` model is spawnned in gazebo the ports opened are not correct

* The FT ports of the arms are opened with /iCub prefix
* The FT ports of the hands are opened with /iCub_fixed prefix
* The FT ports of the legs are opened without any prefx

```
registration name /iCub/left_arm/analog:o ip 10.255.54.112 port 10008 type tcp
registration name /iCub/left_arm/analog:o/rpc:i ip 10.255.54.112 port 10007 type tcp
registration name /iCub/right_arm/analog:o ip 10.255.54.112 port 10010 type tcp
registration name /iCub/right_arm/analog:o/rpc:i ip 10.255.54.112 port 10009 type tcp
registration name /iCub_fixed/left_hand/analog:o ip 10.255.54.112 port 10101 type tcp
registration name /iCub_fixed/left_hand/analog:o/rpc:i ip 10.255.54.112 port 10100 type tcp
registration name /iCub_fixed/right_hand/analog:o ip 10.255.54.112 port 10103 type tcp
registration name /iCub_fixed/right_hand/analog:o/rpc:i ip 10.255.54.112 port 10102 type tcp
registration name /l_leg/left_foot/analog:o ip 10.255.54.112 port 10014 type tcp
registration name /l_leg/left_foot/analog:o/rpc:i ip 10.255.54.112 port 10013 type tcp
registration name /l_leg/left_leg/analog:o ip 10.255.54.112 port 10012 type tcp
registration name /l_leg/left_leg/analog:o/rpc:i ip 10.255.54.112 port 10011 type tcp
registration name /r_leg/right_foot/analog:o ip 10.255.54.112 port 10018 type tcp
registration name /r_leg/right_foot/analog:o/rpc:i ip 10.255.54.112 port 10017 type tcp
registration name /r_leg/right_leg/analog:o ip 10.255.54.112 port 10016 type tcp
registration name /r_leg/right_leg/analog:o/rpc:i ip 10.255.54.112 port 10015 type tcp
```
When the second `icub_with_hands` model is spawnned into gazebo all the FT ports and inertial ports are started with the same names leading to address conflict

NOTE: The same behavior is observed with `inertial` sensor port

Later tried `iCub (fixed)` model, using base `icub` model. In this case also FT and inertial ports are not opened correctly

```
registration name /iCub/inertial ip 10.255.54.112 port 10004 type tcp
registration name /iCub/left_arm/analog:o ip 10.255.54.112 port 10008 type tcp
registration name /iCub/left_arm/analog:o/rpc:i ip 10.255.54.112 port 10007 type tcp
registration name /iCub/left_foot/analog:o ip 10.255.54.112 port 10191 type tcp
registration name /iCub/left_foot/analog:o/rpc:i ip 10.255.54.112 port 10190 type tcp
registration name /iCub/left_leg/analog:o ip 10.255.54.112 port 10189 type tcp
registration name /iCub/left_leg/analog:o/rpc:i ip 10.255.54.112 port 10188 type tcp
[... 33 lines cut ...]
registration name /iCub_fixed_0/head/stateExt:o ip 10.255.54.112 port 10167 type tcp
registration name /iCub_fixed_0/left_arm/command:i ip 10.255.54.112 port 10177 type tcp
registration name /iCub_fixed_0/left_arm/rpc:i ip 10.255.54.112 port 10176 type tcp
registration name /iCub_fixed_0/left_arm/state:o ip 10.255.54.112 port 10178 type tcp
registration name /iCub_fixed_0/left_arm/stateExt:o ip 10.255.54.112 port 10179 type tcp
registration name /iCub_fixed_0/left_leg/command:i ip 10.255.54.112 port 10105 type tcp
registration name /iCub_fixed_0/left_leg/rpc:i ip 10.255.54.112 port 10104 type tcp
registration name /iCub_fixed_0/left_leg/state:o ip 10.255.54.112 port 10106 type tcp
registration name /iCub_fixed_0/left_leg/stateExt:o ip 10.255.54.112 port 10107 type tcp
registration name /iCub_fixed_0/right_arm/command:i ip 10.255.54.112 port 10181 type tcp
registration name /iCub_fixed_0/right_arm/rpc:i ip 10.255.54.112 port 10180 type tcp
registration name /iCub_fixed_0/right_arm/state:o ip 10.255.54.112 port 10182 type tcp
registration name /iCub_fixed_0/right_arm/stateExt:o ip 10.255.54.112 port 10183 type tcp
registration name /iCub_fixed_0/right_leg/command:i ip 10.255.54.112 port 10109 type tcp
registration name /iCub_fixed_0/right_leg/rpc:i ip 10.255.54.112 port 10108 type tcp
registration name /iCub_fixed_0/right_leg/state:o ip 10.255.54.112 port 10110 type tcp
registration name /iCub_fixed_0/right_leg/stateExt:o ip 10.255.54.112 port 10111 type tcp
registration name /iCub_fixed_0/torso/command:i ip 10.255.54.112 port 10161 type tcp
registration name /iCub_fixed_0/torso/rpc:i ip 10.255.54.112 port 10160 type tcp
registration name /iCub_fixed_0/torso/state:o ip 10.255.54.112 port 10162 type tcp
registration name /iCub_fixed_0/torso/stateExt:o ip 10.255.54.112 port 10163 type tcp

```

@traversaro

traversaro, 2018-07-14:
Hi @Yeshasvitvs , was this fixed? Can you leave a line explaining why this issue can be closed? Thanks!

Yeshasvitvs, 2018-07-16:
@traversaro My bad, I thought this issue is a duplicate of [this issue](https://github.com/robotology/gazebo-yarp-plugins/issues/342)

traversaro, 2022-01-13:
Not sure if this is still an issue (@lrapetti ?), but in any case it is more appropriate to have this in icub-models.