robotology-legacy/icub-gazebo-legacy #64: Gazebo 9.0 crashes while trying to visualize links CoM of iCub sdf model

gabrielenava, 2018-04-04:
I tried to visualize the CoM of each link of the [old iCub sdf model](https://github.com/robotology/icub-gazebo/blob/master/icub/icub.sdf). Gazebo 9.0 crashed. This is the error message:

```

gzclient: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:252: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
escalating to SIGKILL on server

```

I don't think this issue is related to Gazebo, in fact I can correctly visualize the links CoM of the Walkman model, see the attached GIF.

[image]

@traversaro @nunoguedelha @fiorisi

traversaro, 2018-04-04:
Cool. I remember visualizing them for this model in old Gazebo versions, so I guess it could be a issue introduced in a new version of Gazebo. Just in case, have you tested the behavior also on Gazebo 9?

gabrielenava, 2018-04-05:
Tested on Gazebo 8 and there are no issues. So it is an issue related to Gazebo 9.

gabrielenava, 2018-04-06:
Issue opened https://bitbucket.org/osrf/gazebo/issues/2454/gazebo-90-crashed-when-trying-to-visualize

gabrielenava, 2018-04-09:
Seem they found the bug:

>I think gazebo is responsible for the crash. I found out that there's a problem getting the bounding box of objects when visualizing CoM. The ignition::math::Box and class's default min and max values changed from zero in ign-math3 to MAX_D in ign-math4. For visuals without any objects attached, it's returning large values which resulted in inf in ogre calculations. I have a fix for it and will create a pull request soon

It will be soon fixed.

traversaro, 2018-08-23:
Gazebo 9.1.0 was released on the first of june, containing this fix (see https://bitbucket.org/osrf/gazebo/src/gazebo9/Changelog.md). Closing.

closed 2018-08-23 by traversaro