robotology-legacy/icub-gazebo-legacy #65: URDF to SDF only loads 4 joints

ahoarau, 2018-09-24:
When I try to convert the generated `icub.urdf` to `sdf` : 
```
gz sdf -p  icub.urdf
```
It only converts 4 joints : 
```
gz sdf -p icub.urdf | grep joint
```
```xml
    <collision name='base_link_fixed_joint_lump__root_link_collision'>
    <visual name='base_link_fixed_joint_lump__root_link_visual'>
    <joint name='l_hip_pitch' type='revolute'>
    </joint>
    <joint name='l_hip_roll' type='revolute'>
    </joint>
    <joint name='r_hip_pitch' type='revolute'>
    </joint>
    <joint name='r_hip_roll' type='revolute'>
    </joint>
```

traversaro, 2018-09-24:
Which model are you trying to convert? 

Issues that could be related: 
* https://bitbucket.org/osrf/sdformat/issues/105/urdf-parser-fails-if-two-consecutive-fixed

ahoarau, 2018-09-24:
This one https://github.com/robotology/icub-model-generator/blob/master/dh/generated/urdf/icubGazeboSim/icub.urdf

Most probably related to this indeed.

traversaro, 2022-01-13:
The reason is actually described in https://github.com/robotology/icub-models/issues/112#issuecomment-1012020159 . TL;DR: Due to problems in the original data used to create the URDF for v1 parts of iCub, `torso_1`, `torso_2`, `l_hip_3`, `r_hip_3` links mass in the URDF was set to 0, and so SDF stops converting (silently, so this could be in a sense a bug  in sdformat) the model as soon as a massless links was reached. As the support/effort is now placed in using the iCub models automatically generated from CAD models, I think we can safely closed this issue.

closed 2022-01-13 by traversaro