robotology-legacy/icub-gazebo-legacy #67: iCub hands inertia tensors

lpenco, 2019-01-14:
I'm making a urdf of the hands of the iCub. I need the inertia tensors for the meshes of the hand. Do you have this information by any chance?

Yeshasvitvs, 2019-01-14:
> I need the inertia tensors for the meshes of the hand

@lpenco I am not sure what you mean by the  inertia tensors of `meshes`. A while back I worked on the [hands urdf](https://github.com/Yeshasvitvs/icub_hands_urdf). I am not certain if the [inertial values used](https://github.com/Yeshasvitvs/icub_hands_urdf/blob/b129da688004a68efffe24bbcd4264b2d5723d6c/l_hand.urdf#L6)  are taken from the CAD models of the hands, but these values along with the collision elements work well in doing simulations in gazebo. 

@traversaro please update on the inertia values used here, which I see are the same values currently present in the [sdf models](https://github.com/robotology/icub-gazebo/blob/fd24d20a3745ce7f549e3cf76557d9a514593c58/icub/l_hand/l_hand.sdf#L5)

traversaro, 2019-01-15:
Hi @lpenco , 
   the only URDF/SDF model of the hands of the robot contained in this repo was added in https://github.com/robotology/icub-gazebo/commit/a8644d447dee94c1e3f88828b3c792fc5e8680af#diff-5edf38f938f0d0dd814e1c52d28289b6 , and was actually extracted from the model available at https://github.com/vislab-tecnico-lisboa/icub-moveit/blob/master/icub_description/urdf/iCub.urdf. If I am not wrong, the inertia for the hands were added by @randaz81 based on some heuristics reasoning and to obtain a stable simulation under Gazebo's default physics settings. 

Regarding inertial parameters, our [single source of truth](https://en.wikipedia.org/wiki/Single_source_of_truth) data related to this are indeed the Creo CAD assembly of the hands available at https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/icub1_icub2/ (see http://wiki.icub.org/wiki/ICub_Project_Mechanical_Documentation_and_Design_Conventions). Note that in this case "source of truth" is just to be intended as a reference, as the inertial properties of the assembly may be wrong if the density or the shape of the CAD model is different for any reason from the final manufactured iCub hand. 

Unfortunately, we do not have a systematic procedure that extracts URDF/SDF of the hands from our Creo CAD assembly, see the issue in https://github.com/robotology/icub-model-generator/issues/29 .

If you are interested in this data, however, we can extract you a Step files of the hands+forearm assembly of iCub from Creo, using the `ap214_is` Step application protocol in Creo, that ensures that the material information is correctly exported (see http://support.ptc.com/help/creo/creo_pma/usascii/index.html#page/data_exchange%2Finterface%2FSupported_STEP_AP203_and_AP214_Classes.html%23 ). Once you have that file, you could use the OpenCascade library (https://www.opencascade.com/doc/occt-7.3.0/overview/html/occt_user_guides__step.html#occt_step_7) to parse the file and extract all the information that you need. 

However, how are you going to use this inertia tensors? The following disclaimer applies: 
* The link of the hands are extremely small, so it is possible that using the exact inertia tensors of the links of the hands, rigid-body simulation using the default physics settings of physics engines such as ODE or DART will incur in numerical errors so bit to affect the stability of the simulation, see https://github.com/robotology/icub-gazebo/issues/22 for a related issue but on the small links of the shoulders. 
* The hand mechanism is extremely compliant due to the use of tendons transmission, so it may be possible that (differently from the case of medium/big robotic arms) the actual dynamics is dominated by friction and compliant effects, and not by rigid body dynamics. In a nutshell, if you are looking for a precise simulation of the hand behavior, perhaps the inertia tensors of the hand are not the first thing you need to check.

I know that @claudiofantacci @giuliavezzani and @xEnVrE are working on this, so they may provide further insights.

traversaro, 2022-01-13:
Hi @lpenco, I will close the issue, feel free to open a new one if you need more info.

closed 2022-01-13 by traversaro