robotology-legacy/icub-gazebo-legacy #68: Gaze controller on gazebo

robertorovella91, 2019-01-21:
So there is not possible to run gaze controller on gazebo ?

_Originally posted by @robertorovella91 in https://github.com/robotology/icub-gazebo/issues/62#issuecomment-456014072_

traversaro, 2019-01-21:
> As regards the `sdf` there is a problem with the way the eyes are placed within the model since their z axis is flipped. The correction is easy and you can find it in this [commit](https://github.com/xEnVrE/icub-gazebo/commit/92680356ac6c21fc8297a30fa8ff6937e94b786b) belonging to a fork of `icub-gazebo`.

If this fix is necessary to align the model contained in this repo with the real iCub robot, feel free to open a PR, thanks!

robertorovella91, 2019-01-22:
Hi @xEnVrE ,

thank you so much, now it works.

traversaro, 2019-01-22:
Hi @robertorovella91 , if it is not a problem I would keep the issue open until the Gaze controller is working fine with the unmodified versions of `gazebo-yarp-plugins` and `icub-gazebo`. 
Did you implemented both the suggestion by @xEnVrE to get it to work?

robertorovella91, 2019-01-22:
Hi @traversaro 
Of Course, yes i changed the sdf file and compiled the plugin.

xEnVrE, 2020-07-17:
We have updates on this issue.

After merging this PR https://github.com/robotology/gazebo-yarp-plugins/pull/492, the gaze controller seems to be running properly on Gazebo.

However, before closing this issue, we still need to address the fact that there is something wrong with the eyes reference frames. Indeed, if you try to increase the eyes vergence it will result in a wrong movement if compared to the real robot (you can check on the following video)

[image]

While https://github.com/xEnVrE/icub-gazebo/commit/92680356ac6c21fc8297a30fa8ff6937e94b786b solves the issue, I am not still sure this is the best solution.

In fact, by looking at the eyes frames in Gazebo

[image]

and comparing them to CAD model 

[image]

it is clear that the sdf model of the robot is wrong.

@traversaro @pattacini @vvasco

traversaro, 2020-07-17:
Small comment: I do not know the logic with which the eye frame have been constructed, but for URDF-derived models it is quite common to have the link frames of the last link of chain **not matching** the actual end effector frame, as the frame of the last link of the chain typically needs to respect some constraint that make it impossible to coincide with the end effecto frame, such as the (URDF) constrain that its origin needs to lie on the axis of the parent joint. That is the reason why if you look at the frame of the `l_hand` link in Gazebo, it will not coincide with the frame usually used by the end effector (that is instead indicated with `l_hand_dh_frame`.

However, I do not think that the mismatch that you see is due to the definition of the eyes frame: apparently what is moved incorrectly is the "forward direction" of the eyes, and that direction is invariant regardless on how you assign the eyes frame.

xEnVrE, 2020-07-17:
I agree with you, the "end effector" frames are not matching and, since nobody is using this SDF with FK libraries or similar for the eyes chain, an equivalent `_dh_frame` frame was never added.

Then, maybe we can proceed by just flipping the axis of motion so that the eyes moves as they should.

traversaro, 2020-07-17:
> Then, maybe we can proceed by just flipping the axis of motion so that the eyes moves as they should.

If that is the problem, then I agree in doing so.

xEnVrE, 2020-07-21:
I think we can probably close this.

traversaro, 2020-07-21:
Don't we need the new limits on the eyes? Or the gaze controller also work well with the old eyes?

xEnVrE, 2020-07-21:
The fact is that, the limits in the current `[LIMITS]` section are, for pure chance I would say, the limits of the coupled joints (I pointed out this in #75). 

Hence, since the current configuration file does not have the limits for the coupled joints, that we started supporting in https://github.com/robotology/gazebo-yarp-plugins/pull/499, and, the controlboard plugin, as a fallback, is using the limits in `[LIMITS]`, to decide the limits of the trajectories of tilt, version and vergence, then everything is working fine.

Maybe, we can complete the update of the configuration files so that everything works because we know that it should work and not by chance :laughing:

traversaro, 2022-01-13:
@xEnVrE is this fixed? 

In any case, I would move this issue to `icub-models` as that one is the repo with supported iCub URDF models.

PR #80 (2020-07-17): Fixes wrong axis directions of joints {left, right}_eye_link
This PR fixes wrong axis directions of joints `{left, right}_eye_link` of robots `icub` and `icub_with_hands`. The robot `icub_fixed` is automatically handled as it is constructed from `icub`.

Fixes #68