robotology-legacy/icub-gazebo-legacy #70: Remove the iCubHeidelberg01 model

traversaro, 2019-04-25:
The iCubHeidelberg01 model present in this repo is not maintained and  it has not been migrated to the new model generation infrastructure at https://github.com/robotology/icub-model-generator . Unless @hu-yue is aware of anyone working on iCubHeidelberg01, I think we can drop it from the repo.

traversaro, 2019-04-25:
cc @diegoferigo

hu-yue, 2019-04-25:
The only person I can think of is @mhubii, if he is not using it, maybe he knows if someone is using it in Heidelberg. 
cc @KevinAStein?

mhubii, 2019-04-25:
I am using it thats true, and it would be cool to have it around until I have finished :)

mhubii, 2019-04-25:
and actually I am not so dependent on whether or not the model can be found here. So if it makes you feel like you could clean up your code by that, do it.

traversaro, 2022-01-13:
This repo has been deprecated some time ago (see https://github.com/robotology/robotology-superbuild/issues/542), so it is not really important to delete this specific model.

closed 2022-01-13 by traversaro