robotology-legacy/icub-gazebo-legacy #78: Use grabberDual instead of grabber in icub camera configuration files

xEnVrE, 2020-07-05:
In files 

https://github.com/robotology/icub-gazebo/blob/master/icub/conf/left_camera.ini

and

https://github.com/robotology/icub-gazebo/blob/master/icub/conf/right_camera.ini

`grabber` is used as the YARP device exposing images to the YARP network.

However, `grabber` is now deprecated in YARP and we should switch to `grabberDual` which is the new default one to avoid issues, e.g. see https://github.com/robotology/gazebo-yarp-plugins/issues/498

If ok, I can open a PR for that.

traversaro, 2020-07-05:
Thanks for the issue, feel free to open a PR, thanks!

traversaro, 2020-07-05:
Just to understand, `grabberDual` is a drop-in replacement for `grabber` or there are some differences the users should be aware of?

traversaro, 2020-07-05:
Related issue: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/100 .

traversaro, 2020-07-05:
Docs: 
* `grabber` docs: http://www.yarp.it/classServerFrameGrabber.html 
* `grabberDual` docs: http://www.yarp.it/classServerGrabber.html

xEnVrE, 2020-07-05:
> Just to understand, grabberDual is a drop-in replacement for grabber or there are some differences the users should be aware of?

Hi @traversaro, I am not that into the topic. It seems that it support two cameras at the same time.

traversaro, 2020-07-05:
I guess @Nicogene can help us answer this question.

Nicogene, 2020-07-06:
> Just to understand, grabberDual is a drop-in replacement for grabber or there are some differences the users should be aware of?

`grabberDual` supersed all the `grabber` functionalities, except for the support also of the audio stream, but recently in YARP the audio management is changed a lot, then I suspect that it doesn't work anymore.
I agree to port the ini to use `grabberDual`, it is the now the default for all the camera devices

traversaro, 2022-01-13:
Fixed by https://github.com/robotology/icub-gazebo/pull/79 .

PR #79 (2020-07-06): Use grabberDual instead of grabber in icub camera configuration files
Fixes #78
closed 2022-01-13 by traversaro